personalrobotics/CCIL #3: Unable to reproduce the PendulumSwingup results

dadadadawjb, 2024-06-29:
Hi team, 

Thanks for sharing the great work! I have tried reproducing the PendulumSwingup experiments, both continuous and discontinuous. I just used the scripts and codes you gave, without any modification. But I find the results do not match the performance shown in Figure 3. (c) in paper [CCIL: Continuity-based Data Augmentation for Corrective Imitation Learning](https://arxiv.org/abs/2310.12972). Do any hyperparameters need to tune, or anything else I need to change to get that results?

[image]

Thanks a lot!

abhaybd, 2024-07-02:
Thanks for your interest in our work! The exact numbers vary depending on the seeds tested, but CCIL should almost always outperform BC on the pendulum task. Please ensure you're running with the hyperparameters specified in the corresponding `.yml` files.

Running the following command on my machine:
```bash
./scripts/train_ccil.sh "pendulum_cont pendulum_disc" "40 41 42 43 44 45 46 47 48 49" 0.0001
```

yields the following results:
```
+-------------------------------------------------------+-----------+---------------+-----------+
| Task                                                  |     Score |   Score (std) |   # seeds |
+=======================================================+===========+===============+===========+
| PendulumSwingupCont-v0_naive                          | -3335.941 |       132.394 |        10 |
+-------------------------------------------------------+-----------+---------------+-----------+
| PendulumSwingupCont-v0_noisy_action_soft_samplingL2.0 | -2527.913 |       363.101 |        10 |
+-------------------------------------------------------+-----------+---------------+-----------+
| PendulumSwingupDisc-v0_noisy_action_slackL2.0         | -2794.500 |       400.131 |        10 |
+-------------------------------------------------------+-----------+---------------+-----------+
| PendulumSwingupDisc-v0_naive                          | -3001.869 |       231.012 |        10 |
+-------------------------------------------------------+-----------+---------------+-----------+
```

As you can see, the exact numbers change due to the associated variance, but CCIL still outperforms standard BC.

dadadadawjb, 2024-07-03:
Thanks for your prompt reply! I got it, but as I observed, especially in the discontinuous Pendulum, the performance between CCIL (-2912.906) and naive BC (-2978.408) is actually hard to distinguish on my machine, even with "40 41 42 43 44 45 46 47 48 49" 10 random seeds. Any good suggestions?

Kelym, 2024-08-06:
Thanks for bringing it into our attention - it seems there are more variance than we initially realized on PendulumDiscontinuous. We validated our config on 10 random seeds and 2 computing machines. We might be able to try tweaking and update the params, if we can find a set of random params that also have similar performance.

In the meantime, do you have a chance to verify the performance on the other task suite? Just want to double check if this is a problem with stochasticity in PendulumDiscontinuous or more.

siddhss5, 2026-10-23:
## Maintainer reproduction contract

This report is now part of [M3: Classic and vector CCIL reproducibility](https://github.com/personalrobotics/CCIL/milestone/4). We will address it through #24 and #25 rather than treating one favorable rerun as resolution.

Closure requires:

- a versioned command and complete resolved configuration;
- the exact environment, dataset, code revision, and evaluation protocol;
- raw per-seed results for behavior cloning and applicable CCIL variants;
- a predeclared statistical summary and uncertainty measure;
- an explicit comparison of continuous and discontinuous Pendulum behavior;
- separation of historical parameters, correctness fixes, and any later tuning;
- links to independently inspectable run artifacts.

The issue may conclude that the discontinuous result is too variable to support a reliable improvement. Closure requires an evidence-backed answer and reproducible artifacts, not recovery of a preferred number.

siddhss5, 2026-10-26:
**Update on how this will be answered (decision of 2026-10-26).**

Rather than rebuilding the original stack to rerun the published scripts, the project measures CCIL's effect on both Pendulum tasks with a rigorous, predeclared study on the maintained implementation (#24). The study:
- uses the same demonstration datasets and the paper's configurations;
- runs 20 paired seeds, with shared evaluation start states and bootstrap confidence intervals;
- compares CCIL against BC and against a tuned NoiseBC;
- adds robustness and demonstration-count analyses.

The predeclaration (in #101) is committed before any run, and the outcome will be reported whatever it is: improvement, no detectable difference, or worse. #25 will publish the report, with a reproducible command and the paper's Fig. 4(c) values for comparison. 'Inconclusive' remains an acceptable answer.

siddhss5, 2026-10-29:
The new analytical Pendulum benchmark in #145 is prospective follow-on work, not an added closure condition for this reproduction report. This issue will still be answered by the maintained-stack evidence in #24/#25 together with the task-aligned interpretation in #143. The response should distinguish the historical objective from the corrected metrics and link #145 as the mechanism-level study that follows, without waiting for that larger study to finish.